ncarandini/KiCad-RP-Pico #2: Wrong classification of pins

bablokb, 2021-03-12:
Your GND and power pins should be classified as power-input.

ncarandini, 2021-04-07:
This isssue is not related with my work that is about the 3D representation of the device. Anyway I'll check if I can fix it on the other libraries. Any help on to do this is much appreciated.

bablokb, 2021-04-07:
I have created a pcb based on your work, see https://github.com/bablokb/pi-pico-devboard-hat. You should find updated versions of the component in the `*rescue.*`-files within the `pi-pico-devboard-hat.kicad`-directory.

ncarandini, 2021-08-05:
Ok done.
[image]

closed 2021-08-05 by ncarandini

raybellis, 2023-08-28:
This seems like it might be incomplete?   The GND at pin38 is still flagged as Bidrectional