ncarandini/KiCad-RP-Pico #4: Change label of pins

bablokb, 2021-03-12:
Not really an issue, more a feature-request/suggestion: change the labels of the pins, e.g. GPIO0/TX

ncarandini, 2021-04-07:
Labels are not part of the 3D Model, I'll investigate on how to change them in the other libraries. thanks anyway for the suggestion. Moreover labels are in the bottom face of the real device, so I'll investigate more also on this.

ncarandini, 2021-08-04:
I've checked the Pin illustration for Raspberry Pi Pico and I dont think is a good idea to change label of pins as you suggested.
As an example, for GP0 we have UART0-TX and SPI0-RX. So I've decided to not change the label of pins.

[image]

closed 2021-08-04 by ncarandini