ncarandini/KiCad-RP-Pico #6: Wrong type for pin 38

SukkoPera, 2021-12-26:
Pin 38 (GND) has _Bidirectional_ type, while I think it should be classified as _Power Input_.

PS: Thanks for your work! :)

ncarandini, 2022-01-09:
Hi, can you further explain? To me GND is GND :-)

SukkoPera, 2022-01-09:
Exactly, but it's [currently set as _Bidirectional_](https://github.com/ncarandini/KiCad-RP-Pico/blob/main/RP-Pico%20Libraries/MCU_RaspberryPi_and_Boards.lib#L45) in your component.

ncarandini, 2022-01-16:
I'll fix it ASAP.

voidloop, 2022-04-09:
Is not pin 36 (3V3) also wrong? I think it should be power output.

SukkoPera, 2022-04-11:
@voidloop: Yes, you are right, that should be fixed, too.

Poofjunior, 2022-05-31:
[image]

Whoop--looks like this is still an issue.