ncarandini/KiCad-RP-Pico #11: Connection between symbol and footprint

asteppke, 2022-06-24:
I followed the installation instructions, added a Pi Pico to an empty schematic and then used the Kicad function "Update PCB from Schematic". Currently this fails with the error message that the footprint under the designation "RPi_Pico:RPi_Pico_SMD_TH" cannot be found. 

The footprint is located in a library called "MCU_RaspberryPi_and_Boards" instead, so editing the symbol to reference "MCU_RaspberryPi_and_Boards:RPi_Pico_SMD_TH" fixes the issue.

impeeza, 2023-01-15:
Use the steps on https://github.com/ncarandini/KiCad-RP-Pico/pull/10 but I get the same result.

The problem is:
If you open the test project, the schematics and the board are displayed correctly:
[image]
[image]
Even the 3dView:
[image]

But if you, by example, add a new <<MCU_RaspberryPi_and_Boards -> Pico Symbol>>, the correspondent FootPrint can not be found:
[image]

The error seems to be what the new "MCU_RaspberryPi_and_Boards.kicad_sym" file, reference the footprint for the symbol as "RPi_Pico:RPi_Pico_SMD_TH":
[image]
and no the value which I think should be: "MCU_RaspberryPi_and_Boards:RPi_Pico_SMD_TH" just like the property for "Pico" on the SCH file of Test project.

If I start a new project and add a new MCU_RaspberryPi_and_Boards -> Pico Symbol, when try to create the PBC, the same error comes:
[image]

if I manually assign the Footprint for the new added symbol
[image]
and then go to create a PBC all goes well.

By the way if you update the Test project PBC, seems the footprint get out of scale, the comparison with the footprint "Connector_PinSocket_1.00mm:PinSocket_1x20_P1.00mm_Vertical"  is huge and the holes do not match:
[image]
Seems the test schematics have the wrong footprint for the connectors, it uses 1.00mm pitch and the "RPi_Pico_SMD_TH" uses 2.54mm (0.10 inches) pitch

I am using KiCAD 6.0.10 in a new installation, Windows 11 fresh installed this morning.

I am missing some step?  my KiCAD is messed up?

thanks for all your effort.

robertknutzen, 2023-01-25:
@impeeza did you ever figure this out? I'm having the same problem

JayM-PT, 2023-02-10:
Same problem here...

HugoCunha, 2024-03-04:
The files provided in this repo are configured as in the author's installation settings.
After importing the symbol, the footprint and the 3D model, as in the [install instructions](https://github.com/ncarandini/KiCad-RP-Pico/blob/main/Install%20instructions.md), you must:

1. Open Symbol Editor:
[image]

2. Double-click on "Pico" symbol (1), and then on symbol properties (2):
[image]

3. In "Footprint" property, click on the value and then on the button to modify it:
[image]
[image]
[image]

4. Click "OK and Save. It should retain the config globally.

tilegg, 2024-06-30:
Thank you so much, @HugoCunha 

Not sure, if this repo is still maintained - do you see any chance of this change making it upstream, @ncarandini?